nam20485/OdbDesign #29: Create documentation

nam20485, 2023-10-05:
## Formats

- [ ] README
- [ ] Wiki

## Content

- [ ] Building from source
- [ ] integration
- [ ] using libraries
- [ ] Overview
- [ ] Architecture

oliv3r, 2024-02-12:
Found this interesting tool, but struggling to see how this is useful, other then the 'it processed the data, now what'. So I think a section about 'how to use' this tool would be interesting.

Though I think #38 would help tremendously. A simple viewer would explain what the purpose was :)

nam20485, 2024-02-28:
@oliv3r Great question!

**The project is a library, so it's intended to be used as part of, or have other applications built on top of it.** 

Some of the value I see the library providing to consuming applications include:

1. Removing the necessity to understand the specific domain knowledge of the ODB++ syntax and how the format is laid out on the filesystem.
2. No other complete (and FOSS) ODB++ parser exists (that I know of).
3. Proprietary organizations routinely expend well over a senior man-year implementing similar solutions for their own use. It's existence can save others a great deal of development time and expertise.
4. Speed/performance- its faster parsing large ODB++ designs than other existing solutions. (the main reason I have seen other orgs implement/re-write their own solutions)
5. Cross-platform- the shared library binaries can be used on all three major platforms (Windows/Linux/MacOS). And the REST API allows the data to be consumed from any platform that can support a REST API client.

Some examples of applications that others have indicated interest in using it in include:

1. A plugin for importing ODB++ design archives in other EDA/ECAD apps (possibly KiCad for instance?)
2. Visual Board Viewers/Editors
3. DRC (design-rule check) tools
4. Tools to evaluate/analyze/optimize PCB board testability and test coverage. Since it creates a high-level abstract logical net-list schematic from the lower level data in the ODB++ (i.e. nets, components, pins, and their connections)

I will add this information to a 'How to use' section in the README.

oliv3r, 2024-03-02:
I did understand the gist of it, but without any 'users' of the library, I couldn't really see how to make heads or tails out of it.

The most confusing part was, that this library can be run in a container, where it seems to be able to transpile files?

A plugin for KiCad of course is what I was searching for, which brought me here. A converter would have been fine as well, heck at this moment even a viewer Would have been eough :)

But what really confused me, is that it was a library (dll, .so etc) but has a REST interface, which makes it sound like a daemon/service really. 

Anyway, Excited to see progress here!

oh-bala, 2024-04-14:
Found this interesting, and considering to create a preprocessor ultizing the library to extract the board information, such as, the minimal space and trace, minimal drill size and etc.

The documents are not yet completed like most of the Pcb opensource tools. Odb++ is getting more and more used on the DFx perspective, so would like to see how this project will keep going. 

@nam20485, Miller, would you create some community channel for actively supporting this project's development?

nam20485, 2024-04-21:
@oh-bala Thank you for your interest in the project!

I have updated the README to include several areas to communicate with me or ask questions about the project and its development.

From the README's [Need Help?](https://github.com/nam20485/OdbDesign/tree/development?tab=readme-ov-file#need-help) section:

- Feel free to message me in the project's Gitter chat room: [odbdesign:gitter.im](https://app.gitter.im/#/room/#odbdesign:gitter.im)
- You can also create an issue: <https://github.com/nam20485/OdbDesign/issues>
- Or you can start a discussion: <https://github.com/nam20485/OdbDesign/discussions>