nam20485/OdbDesign #204: Setting workdir to docker run causes failure

oliv3r, 2024-02-12:
It is quite common to explicitly set the workdir to docker. `--workdir`. However that fails using the dev-741 container.

```
docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: exec: "./OdbDesignServer": stat ./OdbDesignServer: no such file or directory: unknown.
```

Probably the server should be installed in `'/usr/local/bin` and executed explicitly from the path?

nam20485, 2024-02-26:
@oliv3r Thanks for trying the project!

**What is your use case? Why do you need to set the `--workdir` for this container?**

I am not clear on the idiomatic/conventional way to lay out the directory structure for an app such as this. I suppose putting the executable in `/usr/local/bin` would be most correct?

However there are two other directories required by the executable at runtime:

1. `templates/` (where mustache templates for templated HTML endpoints created by the user are found, needs read access)
2. `designs/` (where uploaded ODB++ designs are located and then read from, needs r/w access)

Currently the directory structure is thus:

```
/OdbDesignServer
   /templates
   /designs
   OdbDesignServer (exe)
```

I suppose the two runtime directories should be placed in `/var` somewhere?

_The path for the two directories can be specificed at launch via CLI args. Try `ObdDesignServer --help` or see the [ENTRYPOINT key in the Dockerfile](https://github.com/nam20485/OdbDesign/blob/f9fd60d724d1070d90653c6359e1ed6cb36c8c1c/Dockerfile#L104) for more details._

oliv3r, 2024-02-28:
@nam20485 it's actually considered best practice (depending on how you look at it) at least its part of all my container files and scripts because of this.

I don't need it 'per-say' of course, though I don't think it's "good" for this project to dis-allow it.

The solution of course is also relatively trivial to solve, e.g. putting the expected binaries in /usr/local/bin (or /opt) and setup proper paths and startup scripts. This removes any dependencies on the outside on 'how to setup the container'.

nam20485, 2024-02-28:
@oliv3r That sounds reasonable. I will create a PR with the changes you described.

Do you happen to have a container image that you can link to where I can see any example of the use pattern you are describing?

oliv3r, 2024-03-02:
https://github.com/nam20485/OdbDesign/issues/246 Has said instructions now, so lets close this one in favor of that one.

closed 2024-03-02 by oliv3r