nam20485/OdbDesign #271: some build issues on linux

PinkMushroom, 2024-04-16:
Trying to build this on linux and have found some problems.

First the clone command on the front page

```git clone git@github.com:nam20485/OdbDesign.git```

doesn't work

```$ git clone git@github.com:nam20485/OdbDesign.git
Cloning into 'OdbDesign'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

Looking at that more closely I think it's just using a format assuming that you are accessing using an account ? Just pointing it out because I'm assuming a lot of people, like me, are just going to cut and paste that line ;-)

the standard git clone command using the link generated by github does work

```git clone https://github.com/nam20485/OdbDesign.git```

Then the real problem, the cmake command lines don't work

```$ cmake --preset linux-debug
$ cmake --build --preset linux-release
```

```~/OdbDesign$ cmake --build --preset linux-release
Error: ./OdbDesign/out/build/linux-release is not a directory
```

It sure seems like it expects a linux-release library to be there, which it does not appear to be.  I also searched the tree to make sure it wasn't hidden in a subdirectory and couldn't find it.

I'd really like to offer the solution but I'm not familiar with cmake at all.

Finally, it's not clear if docker is required if you are simply going to write C++ code which uses this library, i.e. Docker isn't needed to build it, is that correct ?

nam20485, 2024-04-20:
@PinkMushroom Thanks for reporting these issues!

>the standard git clone command using the link generated by github does work
>git clone https://github.com/nam20485/OdbDesign.git

I'll change the README to use the working git clone command.

>$ cmake --build --preset linux-release
Error: ./OdbDesign/out/build/linux-release is not a directory

It looks like you didn't run the `$cmake --preset linux-release` command prior to running your `cmake --build ...` command. 

Can you try that and report back if it works or any issues you encounter?

The [full build instructions](https://github.com/nam20485/OdbDesign?tab=readme-ov-file#linux) can be followed from the README.

But please make sure that you have followed the [instructions to setup your build dependencies](https://github.com/nam20485/OdbDesign?tab=readme-ov-file#build-dependencies) first! Or you will encounter some errors about missing dependencies.

>Finally, it's not clear if docker is required if you are simply going to write C++ code which uses this library, i.e. Docker isn't needed to build it, is that correct ?

That is correct. There are two independent options:

1. Build the shared library from source (as you are doing) and include it in your own C++ program.
2. Skip all that and run the REST API server via the docker-compose commands.

I will add something to the README to make that more clear.

nam20485, 2024-04-22:
README updated to resolve these issues also.

PinkMushroom, 2024-04-23:
Thanks for your response.

oops, I had a bit of a reading fail.  I find this confusing from the instructions

$ cmake --preset linux-debug
$ cmake --build --preset linux-release

So the preset builds the `linux-debug` target but then the build/preset builds `linux-release` ?
I would think the first command should be

> cmake --present linux-release

and then further reading fail. Since vcpkg is a microsoft thing, I thought it was needed if you were building under windows.  it looks as though I need to install it for Linux.

You may want to link to the vcpkg github repo.

https://github.com/microsoft/vcpkg?tab=readme-ov-file#quick-start-unix

The instructions going to the microsoft page are not very useful if you are building on linux.

Unfortunately microsoft's instructions for install vcpkg on linux are quite bad.  I'm going to have to admit defeat and stop there.  If you were able to able to install vcpkg on linux I'm really impressed ! lol

> If you don't want to build from source, the built binaries for the shared library are included in the [releases ](https://github.com/nam20485/OdbDesign/releases/tag/v0.9.841)section.

aha, that seems like a better idea than wrestling with install vcpkg.

>   8980848  2024-03-01 14:21   libOdbDesign.so
>   9197976  2024-03-01 14:21   libUtils.so
>    674152  2024-03-01 14:21   OdbDesignServer

so put `lib*.so` in /usr/lib or similar and enjoy OdbDesign programming ?

nam20485, 2024-04-27:
> oops, I had a bit of a reading fail. I find this confusing from the instructions
>
> $ cmake --preset linux-debug $ cmake --build --preset linux-release
>
> So the preset builds the `linux-debug` target [...]

Oops, writing fail! You are correct. I have updated the README to fix that. Good find!

> and then further reading fail. Since vcpkg is a microsoft thing, I thought it was needed if you were building under windows. it looks as though I need to install it for Linux.
> 
> You may want to link to the vcpkg github repo.
> 
> https://github.com/microsoft/vcpkg?tab=readme-ov-file#quick-start-unix
> 
> The instructions on the microsoft page are not very useful if you are building on linux.
> 
> Unfortunately microsoft's instructions for install vcpkg on linux are quite bad. I'm going to have to admit defeat and stop there. If you were able to able to install vcpkg on linux I'm really impressed ! lol

I's not that bad. I distilled the directions down to three steps. You can find them in the Build section of the README now.

> aha, that seems like a better idea than wrestling with install vcpkg.
>
>
> so put `lib*.so` in /usr/lib or similar and enjoy OdbDesign programming ?

Yes, and you need to link your program to the library. Standard shared library linking can be found on the internet.

nam20485, 2024-04-29:
Great! Thanks again for trying out the project.

Let me know if you run into any issues and I can help out.