nam20485/OdbDesign #400: support MatrixFile Color

wen2so, 2025-06-10:
Hi there,

When I tried to use the REST API, I found that there was no color response in `matrix.layer.color`.
I tried to compile it, but there seemed to be an error. Any guidance?

Why is the color attribute commented out in `matrixFile.proto`?

Thank you.

```bash
# error log
/matrixfile.proto:70:22: Expected field name.
"google.protobuf.Color" is not defined.
Import "google/protobuf/color.proto" was not found or had errors.
"google.protobuf.Color" is not defined.
```

[image]

nam20485, 2025-06-10:
I didn't have a good way to model the data structure the ODB++ spec uses for the color when I implemented. I meant to go back and fix it but I have not done so yet. I think the bigger problem was that I didn't how to include the google  standard protobuf Color message. I can look take a look at it and see how much work it would be. 

Can you provide your compile command and environment that resulted in that error message? Were you trying to use the provided protoc compile script?

I take it that the color data is important for your application?

wen2so, 2025-06-16:
comile command:
```bash
$ protoc --version
libprotoc 31.0

$ protoc --proto_path="D:\project\subProject\Protoc" \
         --csharp_out="D:\project\subProject\ProtoBuf" \
         --csharp_opt=file_extension=.g.cs,base_namespace=Odb.Lib.Protobuf \
         matrixfile.proto
```

Are there any suggested ways to use the Color feature with Protocol Buffers? 
Or receive the reponse data with color from Rest API?
Or can we manually convert the color data using strings?

PR #412 (2025-07-08): merge to development
Fixes: #408 
Fixes: #400 

1. fix: protobuf enum conversion for PolygonPart::type
2. feat: support MatrixFile color