nam20485/OdbDesign #463: Support for Parsing tools File in DRILL Layer

Jandody, 2025-08-25:
**Description**
In the current version of the parsing library, it seems that parsing of the tools file inside the DRILL layer folder is not yet implemented.

**Questions**
1.Could you confirm whether parsing for the tools file is already supported?
2.If it is supported, what is the recommended way to access the data provided in this file?
3.If it is not yet supported, are there any development plans for this feature?

If this feature has not been planned, I would be happy to assist with the development or contribute an implementation.

Jandody, 2025-08-25:
I’ve received the invitation, and I’m very happy to accept it. Thank you for the opportunity — I look forward to contributing.

nam20485, 2025-09-08:
> I’ve received the invitation, and I’m very happy to accept it. Thank you for the opportunity — I look forward to contributing.

You are quite welcome. Welcome aboard. I really appreciate the contributions! Top-notch code in the PR, btw. 👍

PR #470 (2025-09-08): Add support for parsing DRILL/tools file and update MatrixFile step/layer parsing
## Summary

This PR introduces support for parsing the `tools` file inside the `DRILL` layer folder and updates the parsing logic in `MatrixFile.cpp` for better step/layer handling.

## Changes

- Implemented **`ToolsFile.cpp` / `ToolsFile.h`** to parse drill tool definitions.  
- Added new proto definition **`toolsfile.proto`** and generated corresponding **`toolsfile.pb.cc` / `toolsfile.pb.h`**.  
- Integrated parsed drill tool data into the existing design model.  
- Updated **`MatrixFile.cpp`** parsing logic: [... cut]
closed 2025-09-08 by nam20485